stuartpittaway/diyBMSv4ESP32 #244: Charge and Discharge preveted?

moscito1010, 2023-09-30:
Ich wusste nicht wo ich das nun schreiben soll weil ich nicht weiß ob es ein bug in dem RS485 Fork ist oder ob ich das Problem bin.
Ich hab hier schon mein Problem geschildert.
https://github.com/ruza87/diyBMSv4ESP32/issues/2#issue-1920389247

Soll ich es hier nochmal schreiben?

stuartpittaway, 2023-11-06:
Was this resolved @moscito1010 ?

moscito1010, 2023-11-06:
Hello Stuart

I'm not sure. Without a memory card, the controller has been running for 26 days without a restart or freezer.
The display is plugged in and MQTT export is activated.
The influxdb function has only been active for 8 days.
The BMS seems to be stable with the last image. I have deactivated the logging function. If logging is activated, the BMS stops reliably after 18 to 26 hours. For me, the conclusion is confirmed that SD cards from the brand manufacturer are no better than the non-name products.

I actually think two things were relevant. SD card and ESP manufacturer.

First of all, I didn't do an OTA flash this time. I did it "like before" :-)
Described externally, deleted everything on the ESP beforehand and then flashed it again.
I got rid of the China ESPs and bought hardware from AZ-Delivery.
Hardware is an ESP32 NodeMCU Module WLAN WiFi Development Board | Dev Kit C V2 with CP2102
https://www.az-delivery.de/products/esp32-developmentboard
Of course the ESP32 Dev Kit C V4 also works. It works in my test system with exactly the same uptime.

Second: I removed the SD card and logged everything via mqtt.

Looking back, the reliability of the SD cards was probably the biggest problem.

moscito1010, 2023-11-06:
Yes, these things are horrible. I find external logging much better. There is a smart home computer in the internal network running a mosquito broker anyway.
A Home Assistant extension would of course be ideal, but I think you've been working on that on the side for a while :-)

Couldn't you also activate a WIKI function here in Github?
That would be an area where workarounds such as: mqtt logging or something similar and point to explanations from your YouTube posts.
I think I would have fun with it :-)

closed 2023-11-22 by stuartpittaway