stuartpittaway/diyBMSv4ESP32 #245: Voltronic and CAN-BUS with WECO or SOL Settings?!

moscito1010, 2023-10-05:
@stuartpittaway 
I learned that some Voltronic models such as the King and MAX series use WECO and SOL (Soltaro) CAN-BUS in the BMS setting.
I checked the hardware on my 3 devices and they all have the CAN-BUS chip integrated in the removable control panel, right next to the RS485 chip.
Source: https://powerforum.co.za/topic/5089-axpert-and-hubble-rs232-ports/?do=findComment&comment=102557

If you are interested I would try to get the protocol information.

moscito1010, 2023-10-08:
@stuartpittaway 
I first looked for the direct route via support and am still waiting for an answer.

Do you know these things? Wouldn't that be a good additional solution for devices with Modbus and/or serial interface?
My Spanish Modbus colleague with whom I exchange ideas bought a “thing” like this.
Basically a tempting idea to work with the dongle. Theoretically this is a translator if I understood correctly.

stuartpittaway, 2023-11-22:
@moscito1010 sorry for the slow reply, didn't realise you had responded.

I doubt that the CAN to RS485 adapter would work without also knowing which registers/protocols you need to "speak" with in the inverter.

So we would still require documentation or working examples of code talking to the WECO/SOL protocol inverters.