stuartpittaway/diyBMSv4ESP32 #246: Reset Daily Amp Hour Count

magpern, 2023-10-09:
I have just upgraded to latest version from a really old version.
The Daily Amp Hour Count is a really wierd value.
[image]

I have tried to reset it with the "Reset Daily Amp Hour Count"-button, but that buttom doesn't do anything.

Processor: ESP32
Version: [169a4ea838986b107de2cf4abc0ed219f89c2bbc](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/169a4ea838986b107de2cf4abc0ed219f89c2bbc)
Compiled: 2023-09-12T10:47:12.660Z

stuartpittaway, 2023-10-10:
Have you updated the current shunt firmware as well?  The two are likely out of sync.

magpern, 2023-10-12:
Update of firmware on current sensor solved problem

closed 2023-10-12 by magpern