stuartpittaway/diyBMSv4ESP32 #248: Failed module communication does not trigger rules on reboot

tobimai, 2023-10-26:
When a module is disconnected, normally the "Internal BMS error" rule is triggered and charging stopped (I have the charger relais programmed on output 4).

But when I then reboot the Controller, it stays in the "Waiting for modules" screen with the outputs in the default state, not triggering the rule despite it being the same "error state" as before. 

Software version is from the 2023-10-14, so pretty new and Hardware is the first version of the board that had the INA chip on-board.

stuartpittaway, 2023-10-27:
Hi @tobimai this is normal behaviour on boot up - the controller will wait for communication with all the modules before normal operation is performed.

What would you prefer the system to do?

tobimai, 2023-11-03:
While it's rather unlikely, there could be a case where the module communication breaks (e.g. broken cable) which turns charging/discharging off (well most setups probably use the outputs for charge/discharge control) as the BMS is unable to monitor cell voltages. 

If the controller reboots now for some reason, all these limitations are gone. So IMO it would make sense if it triggers the "Internal BMS error" rule as long as the controller does not have any valid module data.

stuartpittaway, 2023-11-04:
The trouble with that is as soon as the BMS is rebooted (during an upgrade for example) it will trigger the rule and potentially disconnect the battery/fuse/breaker etc.

tobimai, 2024-03-13:
Sorry kinda forgot this existed. I get the point, maybe a viable solution would be to have a seperate "Boot state" or something like that which it would have while it's trying the establish communication after boot. 

In general it's probably rather unlikely for that to happen, but not impossible.