stuartpittaway/diyBMSv4ESP32 #250: Module V490 (all in one) reports lower cell voltages than actual (parasite voltage reading)

stuartpittaway, 2023-10-30:
The V490 boards self calibrate using a "parasite" voltage detection as per the datasheet.

This appears to be not required and is actually causing incorrect voltage readings to be returned.

Recommend to disable in code.

Fernank2, 2023-11-01:
Hello @stuartpittaway,
The problem continues even when installing this new version on all in one, with the parasite sensor deactivated.

next image, module config
[image]
next image, with active balancer on
[image]
next image, with the active balancer off
[image]

Fernank2, 2023-11-01:
Hi @stuartpittaway, 
The cable from the cell terminal is shared to some connection tabs, from those tabs there is 1 cable for the balancer and 1 cable for the all in one, I doubt that is the problem since it is the same electrical point... Have you tried this configuration like mine?

[image]

The balancer is this:
https://www.amazon.es/dp/B0B9NZVLRS?ref=ppx_yo2ov_dt_b_product_details&th=1

stuartpittaway, 2023-11-01:
> I doubt that is the problem since it is the same electrical point... Have you tried this configuration like mine?

I suspect this is the problem - either that, or there is a problem with cell 0 wire/cable and a higher resistance than the other cables.

I'm using a similar active balancer (which is running all the time at the moment) and don't see the voltage drop you get.

In my most recent video, you can see the installation I have.

Fernank2, 2023-11-01:
Hi @stuartpittaway,
I have checked the resistance and connection of channel 0 and it is correct. The channel 0 cable is the shortest of all.

If I remove the resistance module from the all in one and with the balancer active ON, the graph returns to normal
[image]
[image]
https://github.com/stuartpittaway/diyBMSv4ESP32/assets/143579091/02ea7f70-6107-4609-bff1-17c18c15408b

stuartpittaway, 2023-11-01:
Ah - interesting, do you have any high resolution photos of the passive balancer board (the resistance module) - does this get warm when its NOT balancing?  I'm wondering if there is a faulty MOSFET or poor solder joint on it which is draining the energy from cell 0

stuartpittaway, 2023-11-10:
Hi @Fernank2 any progress on this?

Fernank2, 2023-11-11:
> Hi @Fernank2 any progress on this?

Hi @stuartpittaway ,
I have tried another all in one that @chapulino lent me and I get the same result. The next step is to independently wire all in one and the active balancer, I don't see any logic in the latter but I'm going to try it to rule it out.

Fernank2, 2024-02-07:
Hi @stuartpittaway,
I solved the problem with your first recomendation... independent lines between AIO and active balancer. Now its working correctly. thank you a lot.
[image]

compmaster1, 2024-04-20:
I also have an issue where voltage reading is not the same for each cell. Using ESP32 devc Controller and the 490 AIO on a 12S Li-Ion battery. Each cell is reading off by about 400 to 500 mv. Cell 0 is 3.901 while it reads 3.507, cell 1 is 3.853 while it reads 3.487, etc. No active balancer connects as of now, but I will be using the one from this project once voltages are reading correctly. 
I tried from Modules to configure, manually type in, calculate then save. But on the refresh no change and cannot click anything on cell1-15. If I need to post at https://community.openenergymonitor.org/t/diybms-v4 I can but I had posted in the past and it got messy with keeping up with my issue. My next step is to wire more as different banks but testing with just one for now. I have built 5 controllers and 15 490 AIO boards in preparation/backups.

stuartpittaway, 2024-04-20:
> Each cell is reading off by about 400 to 500 mv.

My guess is that the wrong version of the firmware has been installed (was this board from my shop? I think a batch of them have the wrong firmware).

It's easy to update, take a look at my YouTube video.

compmaster1, 2024-04-20:
This is a JLCPCB right before shop was available. Original FW I used was module_fw_V490_AUTOBAUD_VREF4500_genericSTM32F030K6T6 I have just put on module_fw_V490_AUTOBAUD_VREF4096_genericSTM32F030K6T6.bin with no change in the two. This is from the DIYBMS_Compiled_Firmware_2023-12-27-12-02 folder using an ST-LINK V2 (STM8 & STM32).

stuartpittaway, 2024-04-21:
> This is a JLCPCB right before shop was available. Original FW I used was module_fw_V490_AUTOBAUD_VREF4500_genericSTM32F030K6T6 I have just put on [...]

Ok - do you have a photo of the setup?  The only times I've seen a drastic issue in voltage readings is when the firmware is wrong, or you are sharing the balance wires between the BMS and an active-balancer.

Have you tested the voltage test points on the circuit board (16S) to check the reference voltage is correct (TP5)

compmaster1, 2024-04-21:
All TP look good and TP5 was 4.500 so I put the 4500 FW back on. I have manually soldered Fuse, D1, U3, and U6; other then all the headers. At the BATT header and from 0 to 16 I am getting 48.xxV while in GUI I am getting 43.xxV. I have not connected the passive balance board yet as I am still fabing the aluminum stock, I can test that shortly, I figure I try one step at a time for bassline.

[image]

[image]

stuartpittaway, 2024-04-22:
> At the BATT header and from 0 to 16 I am getting 48.xxV while in GUI I am getting 43.xxV.

Are you running 16 cells?    Are you also connecting the +/- (2 pin header) to the positive/negative of the battery?
Do you have a photo of it installed/connected to the battery?