stuartpittaway/diyBMSv4ESP32 #251: improove VIctron integration change CAN speed to 250 kbps

Bingo2023, 2023-11-01:
current status: CAN is set to 500 kbps
improvement: CAN can be set to 500 kbps or 250 kbps

Background
500 kbps are used to connect diyBMS and VenusOS (speed 500 kbps). Inverter like Multi Plus II is connected via VE.bus (separate communication line e.g. with USB dongel).

250 kbps can be used to connect diyBMS and VenusOS AND new RS Multi Inverter via CAN (speed 250 kbps). In VenusOS you have to lower the CAN speed to 250 kbps (RS Multi and other Victron devices can only do 250 kbps), see attached screenshot. 

[image]

stuartpittaway, 2023-11-01:
Have you tried changing the code yourself to see if this works?

Bingo2023, 2023-11-01:
at a quick search, I could not find the place where CAN in configured. Would you point it point out?

stuartpittaway, 2023-11-01:
https://github.com/stuartpittaway/diyBMSv4ESP32/blob/0f7f03c1c13382ed7031464baa92671fe778afc3/ESPController/src/HAL_ESP32.cpp#L287

Bingo2023, 2023-11-01:
I changed the code and compiled and flasht ---> it works perfect :)
If you could make this change switchable on the Victron/PylonTech setting page would be a real benefit.

[image]
[image]
[image]
[image]

closed 2023-11-28 by stuartpittaway