stuartpittaway/diyBMSv4ESP32 #254: Add Modbus TCP

Chris87-ux, 2023-11-20:
As the controller is already equipped with an Ethernet-connection threw WiFi, it would be great to have an read- out and -in (DVCC values) over Modbus TCP.
I’m actually building my Battery setup, were I have several 16S LiFePo4 Banks in parallel and each do have their separate automatic Fuse, I’m collection the BMS values over MQTT to my PLC, do some internal calculation and emulate to CEON over MQTT the battery array as one.
The collection of all values by MQTT is not that nice for an PLC, it would be great to have the possibility to read out the values over Modbus TCP.

Thanks

stuartpittaway, 2023-11-20:
Are you using an Mqtt broker? The BMS can already publish to that.

What's the final destination for the data? Is it just living on the PLC?

Chris87-ux, 2023-11-20:
Hi @stuartpittaway , sure I'm using an MQTT broker (mosquitto) which is place in an Orange Pi zero 3 which is also acting as an small router to supply WiFi for the BMS. Actually the PLC subscribe to the Topics from the DIYbms’s it is currently working with 2 Battery Banks and 38 topics. Unfortunately the PLS is limited to subscribe to topics, so the 3rd Bank would not work anymore. 
Yes, an alternative would be, to install Node-Red on the Orange pi as well and do all the calculations on it, but my assumption would be, that this is not as reliable as the PLC and all IOs and additional controls can be one on this one as well.

Bingo2023, 2024-01-03:
I'm running Node Red (with MQTT broker) on a Pi 3.
I'm using the Pi-Image from Victron (Venus-OS). It is running very stable since 1 year. I started in the Victron World but left Victron Eco-System. So just Node Red on the Venus is running... I want to stick to it as it is reliable running...

just my thoughts on that.