stuartpittaway/diyBMSv4ESP32 #256: Web Server Authentication

jetronic18s, 2023-11-24:
A great feature would be user authentication in the web server

You could always see all the values or screens of the BMS, but if you want to change something, you have to log in with your username and password.
You could certainly simplify it and only support one user "admin", it shouldn't be complicated with multiple users.

That would be a great security feature, currently the controller hangs unprotected in the network.

Thanks Stuart for the great work!

stuartpittaway, 2023-11-24:
I have considered implementing this previously, however without the added HTTPS/TLS encryption its sort of a half attempt at security, which may give people an over confident feeling of its security posture.

I'll add it to the list though, as I still think it has merit.

jetronic18s, 2023-11-24:
Thank you Stuart.

You could have a look at other projects, e.g. openDTU.
I have no idea how it was solved there.