stuartpittaway/diyBMSv4ESP32 #260: WiFI and MQTT don't automatically reconnect when connection is lost

Roelaynd, 2023-11-27:
WiFi does not automatically reconnect when, for example, the WiFi AP gets restarted. After a power cycle the controller works as expected again.

MQTT does not reconnect when, for example, the broker gets restarted (WiFi stays intact). Other devices on MQTT do reconnect automatically. Have to power cycle the device as the web interface seems to go offline as well?

Don't know if these problems are related but they have the same symptoms. If you like I can do some more testing or capture debug outputs if you can tell me what you would need and give some general direction on how to acquire this.

Some praise:
Love your work Stuart, keep up the good work. Much impressed with the all-in-one module, works like a charm. I have the 4.5V reference installed and 6 of these modules daisy chained, works fantastic. Big improvement on the V4.5 module on handling and accuracy (especially when balancing).

stuartpittaway, 2023-11-28:
See also #239

Roelaynd, 2023-11-28:
Now running the latest build (#237) and at least WiFi does reconnect automatically (tested once so far)

For the record: the software that was on the controller was compiled around the 5th of november somewhere (don't exactly know which build that was).

Will be testing some more today. The issue linked by you appears to be similar indeed. I don't have a screen on the controller but the reception should be of no concern (does not lose WiFi as long as the MQTT broker/WiFi AP is available).

stuartpittaway, 2023-11-28:
Ok, let me know how you get on - are you running a mesh WIFI network?  These seem to give the most issues.

Roelaynd, 2023-11-28:
No mesh network (fortunately). RSSI of -73 (stable connection). Restart of the MQTT broker does not seem to give problems anymore. 

Funny enough, at the first 3 times of taking the MQTT broker down, the controller indicated only after a couple of minutes that the connection was lost (have to manually refresh the integration page for it to show change). At the 4th try it was approx. half a minute and by now it is direct. After rebooting the controller it keeps on showing the change almost immediately.

The count of disconnections starts increasing faster than the count of connections which seems odd but also seems to have no negative consequences.

Roelaynd, 2023-11-28:
Don't know if this is related to the changes made to MQTT and/or if it is intentional, but the new software seems to skip cells when sending the statusses.

It goes from /0/10 to /0/12 for example, skipping /0/11. The next round it might or might not send /0/11 and skip other cells, but there appears to be some pattern to it as some cells seem to be skipped more often than others. As I've got 6 banks of 15 cells, I've now got some cells that are not updated for quite some time whilst their value have changed (both in the last bank, /5/11 and /5/14 at the moment). This happens also when the basic cell reporting is enabled (no difference except for the amount of data send).

RSSI = -71

stuartpittaway, 2023-11-28:
There was a release earlier today to address that, are you running that version?

Roelaynd, 2023-11-28:
That seems to fix the problem perfectly! Messages appear in batches now instead of a continuous trickle which is also better I presume. Thanks a lot for your great work!

If I notice any more trouble on the MQTT or WiFi part I'll notice you, but for now it appears to be stable and able to recover from the various situations that I've thrown at it.

closed 2023-11-29 by stuartpittaway