stuartpittaway/diyBMSv4ESP32 #263: Models page moule numbers not the same as mqtt

rolfbartels, 2023-12-08:
**Describe the bug**
It's very difficult to manage the system when on the home page and modules page the number in bank number and then the module number increments, for example bank 0/0-13 bank 1/14-27 where when the MQTT data is sent it's still showing 0/0-13 and 1/0-13

Would it not make sense to keep banks and modules the way they used to be 
Bank 0/0-x
Bank1/0-x
Bank2/0-x
and so on, 

Which eve way it preferred it would be great to keep consistency, the risk now is if we change the MQTT those that want to retain their history would lose that as 1_0_Volgate become 1_14_Voltage

**Hardware/Software Versions**
Controller version (from PCB): v4.6
Processor: ESP32
Version: [0ee4ed06c281af3a46fbcd0bce816a0d58581436](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/0ee4ed06c281af3a46fbcd0bce816a0d58581436)
Compiled: 2023-11-25T12:02:56.849Z

stuartpittaway, 2023-12-08:
Thanks @rolfbartels which version did you upgrade from?  MQTT has been like it is now for quite some time.

rolfbartels, 2023-12-08:
@stuartpittaway, I keep my system updated as you release updates so it would have been the previous version, but I only notice now as I am trying to troubleshoot am MQTT issue where is seems the controller is not sending all the data, and this is since the Nov update. I have posted a thread in the community first in case I have missing something. So sure, MQTT has probably been like this a while, I only noticed recently.

stuartpittaway, 2023-12-08:
Ok, are you running version https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2023-12-06-09-48