stuartpittaway/diyBMSv4ESP32 #264: Integrated VW ID3 MEB CMU Slave

Michi202020, 2023-12-16:
Hello Stuart,

i used many id 3 Battery and i have also the original CMU from VW and i see that the can bus is hacked.

See here:
https://github.com/anikrooz/housebatt
https://github.com/Tom-evnut/VW-bms/tree/master/VWBMSV2

or GRNbms from Pro-Lox

I can create one more can transresiver and the best is i change the bms between your and the vw id3. I can also pay for your work if you make it.

Best Regards Michael

stuartpittaway, 2023-12-18:
Hello @Michi202020,  are you trying to manage the raw battery cells or integrate with the existing BMS on the battery?

Michi202020, 2023-12-18:
Hello Stuart,

Yes I mean the original vw bms withe can bus 500kbs

See picture. The canbus you can find the link in my first question.
[image]

Michi202020, 2023-12-18:
Sorry again,
For understand I install your controller and if is install than I can say in the softer that I used the 4.4 or 4.9 Board or the vw Id3 board over canbus. That's my idea. Many Thanks and best Regards
Michael

stuartpittaway, 2025-03-30:
I doubt I will integrate this idea at the present time.  Thanks for the suggestion though.

closed 2025-03-30 by stuartpittaway