stuartpittaway/diyBMSv4ESP32 #267: add MQTT Input frame usable external Current-Sensor - enhancement

Bingo2023, 2024-01-03:
I want to experiment in using the bat data from the Charger to "simulate" the current sensor from the BMS.

I'd like to send via MQTT values like:
current, charged-Ah, discharged-Ah, SOC, Voltage

What are the thought on that?
I started already adding fuctions to the code... but seem more difficult as I first expected as we do not have any input functionality at the moment....

stuartpittaway, 2024-01-08:
Hi @Bingo2023 what are you trying to achieve by using the chargers values?  

The BMS only really uses these values to control an external charger (via CANBUS etc.) or to trigger relay rules etc.

Bingo2023, 2024-01-08:
You have developed the "diyBMS-CurrentShunt" to create these number/measurements.

I don't like the idea of cutting&adding HW to the big power lines of the battery.

I try to create a simpler solution as we have all the sensors and measurements already in our systems.
Unfortunately the Pylontech CAN expects data like SOC coming from the BMS....
So the idea is to take these values from the charger and send them via MQTT to the BMS. The BMS sends it via CAN to the charger.

stuartpittaway, 2024-01-08:
But the whole point of the independent current sense on the BMS is to instruct the charger or inverter on what to do.

If you are just going to feed data back into the inverter/charger - what will that achieve?

Bingo2023, 2024-01-09:
"diyBMS-CurrentShunt" sends its data intern/RS485/Modebus to the BMS.

I'd like to experiment with the same idea, using the current sensor from the inverter and sending the values thru MQTT. The currentsensor of the inv will measure exact same as the "diyBMS-CurrentShunt" because it is in the same power-circuit.

stuartpittaway, 2024-01-09:
The state of charge isn't a simple calculation, as you need to accurately measure the energy flow in realtime (thousands of samples per second) - does your charger perform this function?

stuartpittaway, 2024-01-09:
If you have already started messing with the code, you will need to add a lot of MQTT client/input parameters into the controller.

I would also take a look at the on-board current monitoring code and use that as a template for the MQTT logic.