stuartpittaway/diyBMSv4ESP32 #268: Enhancement Proposal for Current Monitoring Overcurrent Rule

lorenzo6201, 2024-01-09:
During my recent exhaustive testing of the controllers, I came across a notable observation regarding the Current Monitoring Overcurrent (Amps) rule. While undoubtedly beneficial, it appears to lack a crucial timing mechanism.

**Issue Description:**
The absence of a timer becomes evident when setting a current value for testing purposes. Upon triggering the rule, I observed that the reset value becomes redundant. For instance, if the trigger value is set to 100A and the reset value to any value below that, the relay experiences continuous cycling. This repetitive disconnect-reconnect cycle places undue stress on the master relay contacts, potentially leading to premature wear.

To mitigate this issue, I propose the incorporation of the following features:

**Timer Functionality:**
Introduce a timer, either user-settable or with a predefined duration. This addition would help reduce the frequent triggering of the rule, preventing unnecessary wear on the master relay contacts.

**Trigger Count Monitoring:**
Implement a mechanism to count the number of times the rule has triggered within a specified timeframe (e.g., 3 times in 60 seconds). If the rule triggers excessively, automatically disable the controller until user intervention. This measure is particularly crucial in preventing potential short-circuits or overcurrent events, thereby averting cable overheating, connector melting, and the risk of fire.

These enhancements aim to not only optimize the performance of the current monitoring rule but also contribute to the overall safety and reliability of the system.

stuartpittaway, 2024-01-09:
Thanks for this @lorenzo6201 as I use a single "pulse" output for over current situations - which trips a manual shunt breaker I've not experienced this issue, but does sound like it needs fixing.

lorenzo6201, 2024-01-09:
> Thanks for this @lorenzo6201 as I use a single "pulse" output for over current situations - which trips a manual shunt breaker I've not experienced this issue, but does sound like it needs fixing.

I have a 500A contactor so in my case is connecting then reconnecting and so on.