stuartpittaway/diyBMSv4ESP32 #270: Dashboard get very crowded with more cells.

Ipirk88, 2024-01-15:
**Hardware/Software Versions**
Controller version (from PCB): Controller V4.6 & AllInOne V4.90
(below can be obtained from the "About" page in the controller web interface)
Version: Release-2023-12-27-12-02

Hello,
is there any way to disable the Cell Voltage for each Cell?

I have 16S 5P Configuration of my Setup (in near future 16S 8P) and the Dashboard becomes very crowded with all that numbers. 

So is there any chance to swith off that feature and only displays the Voltage (and maby also the Temperature) of the cells, when you hover with your curser over it?

[image]

lorenzo6201, 2024-01-15:
Same problem here . I see two ways to fix this : 
1) option to disable the text 
2) a more cleaner look with all details would be to create separate 16S banks one on top of another and  just scroll the page. and would be much easier to read. 

Bank 0  
                 16 cells with temp as before
Bank 1    
                 16 cells with temp
Bank 3 

instead of BANK0     /   BANK 1    /    BANK 2 / etc

Ipirk88, 2024-01-15:
Like the Batrium woud be fine:

[image]

So i agree to the first option - to enable/disable the test.
That woud be nice!

Ipirk88, 2024-01-15:
So please no misunderstanding here - i am going away from Bactrim now (to expensive and SW is lagging sometimes).

But for the "fast overview" it is/was great. You can imidiatly see your peek and low cells i mean its her also (with the collors) but its way to crowded with the text - less Information is sometimes more informations :-)

stuartpittaway, 2024-01-15:
A "narrow" bar graph view without the text is probably the easiest to build rather than splitting the graphs into different banks.

Ipirk88, 2024-01-15:
Splitting it in different banks also will make it not easy for a "quick vielw" - you have to see all at once.