stuartpittaway/diyBMSv4ESP32 #273: Cell Voltage reading freaks out when Active Balancer kicks in - STM32All-In-One

Ipirk88, 2024-01-16:
Hardware/Software Versions
Controller version (from PCB): Controller V4.6 & AllInOne V4.90
Version: Release-2023-12-27-12-02

Hello, 
i charged today my Powerwall full the first time with diyBMS. 

The Cell Voltage reading is stunning accurate.
But i discovered an error.

When the Active Balancer (NEEY 4A Active Balancer) kicks in, the Cell voltage reading freaks out (see in the picture - hard to catch, because it changes so fast).

You can see sometimes it measures verry low Voltage and sometimes it measures realy high Voltages - and than it measures the right Voltage. 

I think the Switching Converter inside the NEEY makes some noice on the Cable - and the BMS detects it. 

I head a "amateur" look into your Code and i discovered that you only takes one Cell Voltage reading per round. 
Is it not better when you take several readings and filter them a bit - so that we are not so influenced by any noise on the line?

[image]

stuartpittaway, 2024-01-16:
Don't share the balance cables, when the balancer is pulling a large current it will ALWAYS affect the DIYBMS cell reading

Ipirk88, 2024-01-17:
OK - tested yesterday with the Cap´s (1000µF) parallel to the Battery Cell (mounted on the Therminal Block) and the results are the same.
I ordered yesterday cables an Crimp Connectors for connecting the BMS directly on the Battery Cell - i let you guy´s know the result (Material shoud be her on Friday).
But thanks so far for your support - and of corse for the great project!!
BR Ingo

Ipirk88, 2024-01-20:
OK - Tested it today with new Balancer Cables - Active Balancer and AllInOne diyBMS have now theyer own Cables to the battery terminals. 
And it works now - now more interferance!

I also measured with the oszilloscope - the Current is in range (4A max - so no Current peaks). But the Active Balancer makes some noice on the Cables - and that is what the BMS detects. 

Thanks for your help!!!

stuartpittaway, 2024-01-24:
Can this issue be closed now?

Ipirk88, 2024-01-25:
Thanks - yes can be closed!

To wire the Active Balancer and the diyBMS seperat to the battery terminals is the solution!

closed 2024-01-25 by Ipirk88