stuartpittaway/diyBMSv4ESP32 #274: Temperature display 16S balancer

gartenjoe, 2024-01-29:
Hello everyone. I have a strange phenomenon. When I plugged in the balancer module, every second cell shows minus temperature.
This is the case with 2 modules on the same controller. 3 modules are working properly. Since there is hardly any electronics on the balancer module, I don't understand this error. Does anyone have an idea? Question to Stuart: How is the temperature actually determined via the balancer module? greeting Detlev
[image]

stuartpittaway, 2024-01-29:
> Stuart: How is the temperature actually determined via the balancer module? 

Hello @gartenjoe the temperature is measured using on-board thermistors, marked TH1 and TH2 on the balancer module. 

Are you using external cell temperature sensors as well ?  (connected to T1/T2/T3) 

Are you using the latest code on the V4.90 board?

gartenjoe, 2024-01-29:
Yes, I also use T1/T2/T3. It's not the latest code, but all 5 modules have the same code.

stuartpittaway, 2024-01-29:
Ok, there was a change recently for issues with temp sensor reading - can you update the modules to the latest version please?

gartenjoe, 2024-01-29:
Hello. Error is found. One of the two NTCs on both modules was broken. Thanks to Stuart and have a good time.

stuartpittaway, 2024-01-29:
Oh wow, did JLCPCB build these for you?

closed 2025-03-30 by stuartpittaway