stuartpittaway/diyBMSv4ESP32 #275: Suggestion improve Cell runaway function

red0909, 2024-01-31:
at the moment the 4.9 all in one module balances only one cell after an other, but it should balance all cells at same time if they are over the set voltage and deviation.

Bingo2023, 2024-02-05:
I used last summer the Victron CAN & Victron HW and it worked great.

[image]
here are the settings check especial the red marked.
Also make sure that the Charger Voltage is little higher to reach that levels. 
I dont remember the bypass voltage exactly but try 3,65V +/- little 2-5mV - I was able to find good settings.
Below is a datasheet from my CATL cells, they propose settings for the BMS (1st Level 3,8V 2nd 4,0V).
Check your datasheet, but maybe 3,5V is too low. The higher the more they tend to run away. If you go like 3,5V they are all very same level -> all cells will balance at same time and each cycle.

[image]

red0909, 2024-02-05:
3,5V is optimal its enough for balancing, but your method does not work for me look the image above
the bypass voltage is never reached and if its reached, if i set it at or below charge voltage, it will constantly burn energy in summertime.
this is not designed for that constant bypass, but the runaway function stops burn energy if deviation is under the set value so it will not burn energy constantly in summer.