stuartpittaway/diyBMSv4ESP32 #278: CVL and Venus bug

buddhafragt, 2024-02-14:
I used the latest software  9a0a3ed26f2dc973d3faf3e2799a6688da3ab13e and notice the Victron MPPTs (I use 2) have different charging power, but should have absolutely the same. So I did some research and monitor in Venus/Node-Red the CLV Voltage. 
Venus show 51.2V, but the charging voltage is set to 53.0V, also the float Voltage is set to 49.5V. Here seems to be something wrong....
[image]

stuartpittaway, 2024-02-15:
You have dynamic control enabled - this will automatically reduce the charge voltage to prevent the **cells** going over-voltage.

The float voltage is only used once the cells reach 100% state of charge - it will then enter the absorption timer (60 minutes) then drop to float.

The screenshot given shows a cell target of 3.45V = 16 x 3.45=55.2V maximum voltage.  This will be limited by the 53.0V maximum you have set.

buddhafragt, 2024-02-15:
So the Software also regulate the voltage? I thought, only the current is regulated according to the curve.
I see, if I switch dynamic off, it goes to charge voltage. 
The voltage is correct, bcs. I have only 15 cells!
Thanks for the explanation. I think, we need a manual...

stuartpittaway, 2024-02-15:
Yes agreed, documentation!

closed 2024-02-15 by stuartpittaway