stuartpittaway/diyBMSv4ESP32 #279: Loosing wifi on reboot from 5v board power but not usb

dougnutz, 2024-02-18:
**Describe the bug**
tldr; If the usb plug has power the wifi connects fine on reboot. If the controller is powered solely from a dc/dc converter via 5v input the wifi will not connect, wifi.json will be erased, and the screen will show "configure wifi"

**Hardware/Software Versions**
Controller version (from PCB): 4.5 w v4.9 all in one monitoring board
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: [9a0a3ed26f2dc973d3faf3e2799a6688da3ab13e](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/9a0a3ed26f2dc973d3faf3e2799a6688da3ab13e) 
Compiled:  2023-12-27T11:59:52.391Z

**Software built from master and reproduced with [latest](https://github.com/stuartpittaway/diyBMSv4ESP32/releases/download/Tag-2023-12-27-12-02/Compiled_Firmware_2023-12-27-12-02.zip)**

Board purchased  [ESP32-DevKitC](https://www.amazon.com/gp/product/B09KLQF4RR/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1)

**To Reproduce**
Steps to reproduce the behavior:
1. Complete setup and save wifi from the storage and logging page
2. disconnect USB
3. reboot
4. notice the blue wifi config screen

Things done so far.
1. The issue was noticed from the first setup, using two separate but identical builds.
2. Configure wif via usb, 
3. save wifi.json. 
4. Shut down remove sd card, 
5. verify wifi.json
6. reassemble
7. boot via 5v
8. observe blue config screen
9. shut down
10. remove sd card
11. observe no wifi.json
12. copy wifi.json to sd card from computer
13. reassemble but do not connect to the battery bank (no 5v no sense wires)
14. boot from usb
15. system comes up fine, wifi connects
16. reboot (usb still connected)
17. system boots and connects to wifi
18. connect to battery 
19. reboot with usb and battery connected
20. system boots and connects to wifi
21. remove usb
22. reboot
23. system goes to blue wifi config screen
24. shut down, and disconnect
25. remove sd card and verify wifi.json is missing

The issue was reproduced without the shunt add on board and later with the shunt add on board installed. No change

Here is a sample of the serial output
```
No key press detected
I (5489) diybms-set: Load WIFI config from FLASH - return 1
I (5489) diybms-set: IP=0,GW=0
I (5489) diybms: Wifi Config: SSID:, Manual Config:0
I (5491) diybms: Manual IP:0.0.0.0, Netmask:0.0.0.0, GW:0.0.0.0, DNS1:0.0.0.0, DNS2:0.0.0.0
I (5500) diybms: Checking for /diybms/wifi.json
E (5507) diybms: Error deserialize JSON
I (5508) diybms-set: Load config
I (5511) diybms-set: Count: UsedEntries = (250), FreeEntries = (380), AllEntries = (630)
I (5546) diybms: homeassist_apikey=tmkTKWQdAxit#SJQqjD8*n#R
I (5547) curmon: Chip=INA229, Revision=1
I (5647) diybms: Onboard/internal current monitoring chip available
I (5647) curmon: RSHUNT=0.00033333, CURRENT_LSB=0.00028610, R_SHUNT_CAL=5000
W (5649) curmon: WARNING: True Max Current Measurable 122.88 Amps
I (5655) curmon: RECALC: RSHUNT=0.00033333, CURRENT_LSB=0.00023437, R_SHUNT_CAL=4096
I (5664) diybms-hal: CAN driver installed.  Filter=0 Mask=4294967295
I (5668) diybms-hal: CAN driver started
D (5673) diybms: Setup RS485
D (5675) diybms: Configure RS485
I (5680) diybms: ** Controller changed state from PowerUp to Stabilizing **
D (5685) diybms: starting wifi_init_sta
I (5709) diybms: WIFI SSID:
I (5789) diybms: Hostname: DIYBMS-00C7BD7C
D (5790) diybms: wifi_init_sta finished
I (5790) diybms: TFT screen is INSTALLED
I (12043) diybms: WIFI_EVENT_STA_START
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x300a (ESP_ERR_WIFI_SSID) at 0x4008ea0b
file: "src/main.cpp" line 1616
func: void event_handler(void*, esp_event_base_t, int32_t, void*)
expression: esp_wifi_connect()
I (12115) diybms-tft: Wake up screen
D (12166) diybms: total_free_byte=131460 total_allocated_byte=161928 largest_free_blk=94196 min_free_byte=131260 alloc_blk=383 free_blk=3 total_blk=386
I (14938) diybms-rules: Set error 4:WaitingForModulesToReply
I (17433) diybms: Task 1
I (19948) diybms-rules: Set warning 9:NoExternalTempSensor
I (19948) diybms: Active errors=1
I (19998) diybms-tft: Wake up screen
I (22935) diybms: Task 2
```

note the error `diybms: Error deserialize JSON`

dougnutz, 2024-02-20:
I agree with you, it seems likely that I've got something "off". I wonder if it's common to use the 5v board supply. Digging into the code a bit I suspect some sort of timming issue or false positive in the Space key  ckeck during startup. Are there any other paths to the blue config wifi screen than pressing spacebar?

> Can you check which 3.3v regulator is installed on the esp32 board? 

I'll check the regulator, but off hand I used the BOM from the repo.

> If you flash the esp32 using the pre built image from GitHub do you get the same problem?

Yes I tried "latest"

I also tried an over the air update, which also reproduced the problem.

dougnutz, 2024-02-20:
Am I correct in thinking the 3.3 reglator is in position U6?
From the [BOM](https://github.com/stuartpittaway/diyBMSv4/blob/master/ControllerCircuit/ComponentListBOM.md)

[image]

From my order history

[image]

dougnutz, 2024-02-20:
I just realized my board is v4.6 not 4.5. The readme indicates 4.5 is the latest but my board has 4.6 on the silkscreen. I don't think that matters for this issue.

stuartpittaway, 2024-02-21:
> Am I correct in thinking the 3.3 reglator is in position U6?

No - the 3.3V regulator on the actual ESP32 board, normally in the centre, just above the flash memory chip.

dougnutz, 2024-02-21:
I ordered two boards from a different seller to see if it's related to the particular board.

dougnutz, 2024-02-23:
This issue doesn't reproduce with the new ESP32 board. 
I think you can close this issue, the board that was problematic is linked in the discription. 
Thanks!

stuartpittaway, 2024-02-23:
Ok - how strange, wonder if its a faulty flash chip on the ESP32?

closed 2024-02-23 by stuartpittaway