stuartpittaway/diyBMSv4ESP32 #280: Dynamic Charging

buddhafragt, 2024-03-05:
I still have a problem with the Dynamic Charging setup.
When it is on, DIYBMS adjusts CVL down (for no reason that I can see), CCL goes down according to the curve, but CLV (51.4V) is below the maximum charge voltage (52.5V) even though cell voltages are below "Charge target cell maximum voltage".
Something else is wrong here I think.
[image]
[image]

stuartpittaway, 2024-03-06:
These are my settings for a 16S bank of EVE 280K LFP cells.

I think its the "voltage sensitivity" setting which is making the charge slow down too quickly.

Ultimately the dynamic charging formula attempts to never let a cell get near a "dangerous" level - so if 1 cell begins to peak, the whole charge voltage will drop to prevent that cell going higher.  Normally at this point, other cells catch up and/or an active balancer will reduce that peak so normal charging can resume.

As you appear to have a "weak" cell (number 7) I would suggest trying to charge that one cell up individually (using a bench power supply for example) to get a better balance.

[image]

stuartpittaway, 2024-03-13:
Hi @buddhafragt did you make any changes to your setup?  Any improvements?

buddhafragt, 2024-04-12:
Hi @stuartpittaway , I increase the Battery charge voltage to 3.55V/cell, so it is working better!
A big problem is also from Victron, bcs. the MPPTs calculating the charge current wrong. The Victron system is not able to set the battery charging current correctly because the consumption current is not measured accurately. (MPPT current - consumption current = battery charging current)

stuartpittaway, 2024-04-15:
> A big problem is also from Victron, bcs. the MPPTs calculating the charge current wrong. The Victron system is not able to set the battery charging current correctly because the consumption current is not measured accurately. (MPPT current - consumption current = battery charging current)

I don't know if this is a problem with Victron setup or not - I don't really have much information about Victron configuration.  Perhaps ask on the OpenEnergyMonitor forum for help?