stuartpittaway/diyBMSv4ESP32 #281: Use 16 Cell chip but change script to 12 Cell

Michi202020, 2024-03-09:
Hello Stuart,

i change your design for my id3 battery. I used the MAX14921ECS for 16 cell but i make what do you say withe the bright from 12 to 16Cell but on the most time the system say he found 13 cell but number 13 is only 0,005V. Can i change the scipt that he used only 12Cell no more?

Best Regards
Michael .G

Michi202020, 2024-03-09:
[image]
[image]
[image]
[image]
[image]
This is my ID3 Version.

stuartpittaway, 2024-03-09:
Have you connected the unused cell monitoring pins on the max chip to the highest cell voltage? 

You can also just hard code the number of cells in the code if you wish.

Michi202020, 2024-03-09:
Hello Stuart,
Yes I have connect from cell 16 up to 12 but the first pcb make it good and the next I have this problem. For me it's better I hard code this but I don't understand where.
I test it withe max cell but after this I have  o connection to my controller. Can you say me on what position I must change for the 12 cells?

Best Regards
Michael

stuartpittaway, 2024-03-11:
If you remove these lines from main.cpp of the all-in-one source code...

https://github.com/stuartpittaway/diyBMSv4ESP32/blob/9a0a3ed26f2dc973d3faf3e2799a6688da3ab13e/STM32All-In-One/src/main.cpp#L533

```
  // Sometimes on power up, the chip returns 1 cell more than actually connected, so take a few
  // samples and return the lowest cell count
  number_of_active_cells = 255;
  for (size_t i = 0; i < 4; i++)
  {
    auto x = queryAFE();

    if (x < number_of_active_cells)
    {
      number_of_active_cells = x;
    }
  }
```

and replace with

```
queryAFE();
number_of_active_cells = 12;
```

Michi202020, 2024-03-11:
Hello Staurt,
its working. Thanks for your help and i can finish my work.

Best Regards
Michael G.