stuartpittaway/diyBMSv4ESP32 #284: Possible to send 'wrong' voltage info over the CANBUS to inverter ?

deekoowee, 2024-03-13:
My inverter is PylonTech compatible, but its internal software do not accept dynamic charging control.
Inverter does not have posibility to change any of the battery details.

I can read the inverter's modbus data and found that it shows the battery voltage and soc CANBUS data from from DiyBMS.
I guess, that inverter use canbus voltage and soc to control the charging/discharging.
 
The inverter work between 48-55,8V. I would like to charge a little more, say 57,4V = more kWh storage.
I have 14S13P Leaf battery pack.

I'm not familiar with coding and I don't know, how I should change the code.
I wish like to try to cheat the inverter and test this workaround. 
When real battery voltage is 57,4V, CANBUS send 55,8V.
So it is simply deduct 1,6V from all measured voltage and send the cheat voltage data over the canbus.

Any ideas ?

deekoowee, 2024-03-13:
I agree, that there might be compatible issue with Pylon protocol.
Running now Pylon emulation and selected Freedom Won; Freedom Lite battery.
It goes higher (55,8V) than Pylon batteries with Pylon emulator

Supported batteries:
BYD; BatteryBox
Dyness; Power Box
Forsee; Rack 48-3, ZenHome
Leclanche; Apollion Cube 
Pylontech; 3000C, Force L1, Force L2, Phanom-S, US2000, US2000B, US2000B Plus, US2000C, US3000, US5000 
UZ Energy; L051100-A
Unicorn; EnergyCube 850S
Weco; 4K4, 5K3
Yuasa; LIM40E-13, LIM40E-14

stuartpittaway, 2024-03-13:
In diybms have you tried setting the inverter to DEYE?  Your inverter may have the same problem that one has

deekoowee, 2024-03-15:
DEYE setting didn't make any change to top voltage.

stuartpittaway, 2024-03-15:
Ok, looks like a limitation of the charger/inverter you are using.

closed 2025-01-02 by stuartpittaway