stuartpittaway/diyBMSv4ESP32 #285: Shunt Voltages over 40.96mV can't be measured when using internal shunt

tobimai, 2024-03-14:
ADCRANGE is set the 1, which means the INA229 can measure a maximal shunt voltage of 40.96mV. As you can set shunt voltages larger than this this seems like a bug and is confusing if you don't know how the INA works. 

So IMO either a manual or automatic toggle between ADCRANGE 0 or 1 is needed, options over 40mV have to be hidden when using the internal INA or at least there should be a out-of-range warning.

tobimai, 2024-03-14:
I actually just can't read and didn't notice there is a warning about that on the settings page. 

Still, maybe some visual red border when out of range or something like that would be interesting (or a way to support ADCRANGE 0)

closed 2024-03-14 by tobimai

stuartpittaway, 2024-03-15:
This code was submitted
https://github.com/stuartpittaway/diyBMSv4ESP32/issues/214

but I don't have any way to test this out as all my shunts are 50mV.

Maybe adding a "maximum current sensor reading" value/alert would be a good idea as you mention.