stuartpittaway/diyBMSv4ESP32 #287: CAN - Failed to queue CANBUS message

brii999, 2024-03-24:
Hi.

First of all, thanks a lot for creating such a great project like diyBMS.

I am assembling my own energy storage and I have reached the stage where I want to connect diyBMS with Cerbo GX from Victron.

I created the appropriate cable, checked whether it had continuity and connected it to Cerbo and BMS. Unfortunately, the data from the BMS did not appear in Cerbo and the BMS itself started to restart.
I connected via the serial port and watched the logs:

```
..............

No key press detected
I (6374) diybms-set: Load WIFI config from FLASH - return 1
I (6375) diybms-set: IP=0,GW=0
I (6375) diybms: Wifi Config: SSID:XXX, Manual Config:0
I (6378) diybms: Manual IP:0.0.0.0, Netmask:0.0.0.0, GW:0.0.0.0, DNS1:0.0.0.0, DNS2:0.0.0.0
I (6386) diybms-set: Load config
I (6389) diybms-set: Count: UsedEntries = (249), FreeEntries = (381), AllEntries = (630)
I (6425) diybms: homeassist_apikey=UsMI64fQVWqCGxbBB2vKajGv
I (6425) curmon: INA229 chip absent
I (6426) diybms: Onboard/internal current monitoring chip not installed
I (6430) diybms-hal: CAN driver installed.  Filter=0 Mask=4294967295
I (6435) diybms-hal: CAN driver started
D (6440) diybms: Setup RS485
D (6442) diybms: Configure RS485
I (6447) diybms: ** Controller changed state from PowerUp to Stabilizing **
D (6451) diybms: starting wifi_init_sta
I (6475) diybms: WIFI SSID: XXX
I (6547) diybms: Hostname: DIYBMS-00EC889C
D (6547) diybms: wifi_init_sta finished
I (6547) diybms: TFT screen is NOT installed
I (13594) diybms: WIFI_EVENT_STA_START
D (13694) diybms: total_free_byte=161052 total_allocated_byte=132712 largest_free_blk=110580 min_free_byte=159108 alloc_blk=360 free_blk=7 total_blk=367
I (14402) diybms: WIFI_EVENT_STA_CONNECTED channel=1, rssi=-53
I (14414) diybms: IP ADDRESS HAS CHANGED
I (14415) diybms: Request time from time.google.com
I (14416) diybms: Timezone=UTC0DST0
I (14417) diybms: The current date/time is: Sun Mar 24 15:46:07 2024
I (14442) diybms: You can access DIYBMS interface at http://DIYBMS-00EC889C.local or http://192.168.99.210
E (14592) diybms: Failed to queue CANBUS message (0x103)
I (14593) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:128, arb-lost:0, bus-err:14, state: OFF / RECOVERY NEEDED
W (14598) diybms: Initiating recovery
E (14601) diybms: Failed to queue CANBUS message (0x103)
I (14606) diybms: CAN STATUS: rx-q:0, tx-q:0, rx-err:0, tx-err:0, arb-lost:0, bus-err:14, state: STOPPED
I (14615) diybms: Starting CANBUS ESP_OK
W (14864) diybms-webfuncs: Incorrect cookie rec 5d*gLZjZyyDXVnXi2gF6
E (15720) diybms: Failed to queue CANBUS message (0x103)
I (15720) diybms: CAN STATUS: rx-q:0, tx-q:3, rx-err:0, tx-err:128, arb-lost:0, bus-err:14, state: OFF / RECOVERY NEEDED
W (15725) diybms: Initiating recovery

assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0)

Backtrace: 0x400841a1:0x3ffbf52c |<-CORRUPTED

ELF file SHA256: 31418cd666101b8d

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    62][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (66) diybms: 
[... 481 lines cut ...]
I (47595) diybms: Task 2
W (47892) diybms-webfuncs: Incorrect cookie rec QdIbLi78xhvaTtGitMv6
E (48456) diybms: Fail to queue CANBUS message (0x103)
E (48556) diybms: Fail to queue CANBUS message (0x103)
E (48556) diybms: Fail to queue CANBUS message (0x103)
E (48556) diybms: Fail to queue CANBUS message (0x103)
E (48559) diybms: Fail to queue CANBUS message (0x103)
E (48564) diybms: Fail to queue CANBUS message (0x103)
E (48569) diybms: Fail to queue CANBUS message (0x103)
I (49234) diybms-rules: Set error 4
I (49235) diybms-rules: Set warning 9
I (49235) diybms-rules: Set warning 7
I (49235) diybms-rules: Set warning 8
I (49237) diybms: Active errors=1
E (49674) diybms: Fail to queue CANBUS message (0x103)
```

I tried using the second and third controller boards and different ESP32 - it didn't change anything, still the same messages. I tried to look for some trace on the network, but I only found one to check the voltage level on PIN 8 of TJA1057GT/3J - in my case it was 0V regardless of whether CAN was turned on or off.
I also decided to check the supply voltage for this system and it was 5.1V and for Vio 3.2V.

The 120 Ohm resistor is active (JP1 bridge soldered), the resistance between the terminal ends has been checked. Cerbo has a 120Ohm terminator. To make sure that the problem was not Cerbo, I checked using Sparkfun CANBUS Shield whether any packets were passing through. After connecting to Cerbo, you can see that Cerbo is sending packets and I am receiving them without any problems. I later connected to DIYBMS to the CAN terminals, but unfortunately there were no packets visible here and the CAN send error counter kept growing. I replaced the pins just in case, but it didn't change anything.

The CAN chip I have soldered (the entire assembly was done by JLCPCB) is NXP A1057G3 (CQSL4 Zn206).

I have no idea what could be wrong - I would be grateful for any tips on how to pursue the issue further.

stuartpittaway, 2024-03-25:
Hello @brii999, thanks for trying out DIYBMS.

The CAN controller chip I use on the controller is SN65HVD1050D https://www.ti.com/lit/ds/symlink/sn65hvd1050.pdf

This chip operates at 5 volt and doesn't use pin 5 - its not connected on the PCB.

If you are using the TJA1057GT/3 then pin 5 is expected to be "VIO" - you might be able to make your chip work by connecting pin 5 to +5V (a wire soldered between pin 3 and pin 5 should do it).

Alternatively, swap out that CAN chip for the recommended SN65HVD1050D

brii999, 2024-03-26:
Thanks for reply. I tried with providing 5V supply for PIN5 of CAN controller with no luck. I ordered mentioned chip (SN65HVD1050D). When it arrives I'll solder it and provide update in this topic. Once again thanks for interesting in this problem.

stuartpittaway, 2024-03-28:
Its a fairly simple circuit - first check that your ESP32 is the correct pin layout (devkitc footprint).

After that, check the connection to these two chips at U1 and U7.  U1 is tiny, so easy to cause solder bridging.

Also check that your CANBUS A/B wires are correct.  You can also try without a ground wire between controller and charger/inverter.

[image]

brii999, 2024-03-28:
I have no TSX0102DCU chip - TXD and RXD of SN65HCD1050D goes stright to ESP32 board to pins:
TXD (1) on SN65 to G16 on ESP32 (U2_RXD)
RXD (4) on SN65 to G17 on ESP32 (U2_TXD)
Revision of diyBMS 4.4 board is: c8f1403

atanisoft, 2024-03-28:
If your PCB is [this](https://github.com/stuartpittaway/diyBMSv4/blob/c8f14032669a9fdfd0f7e264b615031e2548635f/ControllerCircuit/export/ControllerCircuit-top.png) one, you should be able to use the TJA1057/3 chip by making sure JP5 is bridged to 5v. I've not used mine diyBMS boards for CAN but I do have extensive use of TJA1057/3 (and a few others) in the split voltage configuration without issues.

Also the error code 0x103 indicates that the TWAI/CAN driver is in an incorrect state for sending, this is usually due one of the following:
1. CAN-RX and CAN-TX configured in reverse.
2. CAN-H and CAN-L reversed on one of the devices.
3. Missing 120Ohm termination on *BOTH* ends of the CAN bus.

A very quick test for the third item, turn off all devices but leave the CAN bus connected on both ends. Set your multimeter to resistance testing and check for a 60Ohm resistance (roughly) between CAN-H and CAN-L. If you see 120Ohm then only one side is terminated and you might be able to fix this by soldering JP1. If you see a dead short between CAN-H and CAN-L check the soldering of the TJA1057/3 IC to make sure there isn't a stray bridge, you can see the traces are pretty much direct from U1 (TJA1057/3) through the center two pins on the right side (6 and 7) connecting to D8 (CAN TVS) and then to J7 CAN-H/CAN-L. CAN-H is on pin 7 of TJA1057/3 and is connected to R6 and D8 (top left) -> J7. CAN-L is routed to JP1 -> D8 (right side) -> J7.

brii999, 2024-03-30:
Thanks for reply in this topic.

1. I'll check this one more time
2. I tried both options of connection with the same result
3. I've got solder blob on JP1 and resistor in plugs connected to sparkfun board. Resistance between lines is exactly 60,05Ohm (i'am not joking - its practical ideal).

I have no idea whats can be wrong...

stuartpittaway, 2024-03-31:
> I have no TSX0102DCU chip - TXD and RXD of SN65HCD1050D goes stright to ESP32 board to pins:

Sorry, my mistake, if you have the older PCB board, then the TJA chip is the correct one to use.

brii999, 2024-04-20:
A little udate: I took two controller boards and two ESP32, wrote small programs - one to send data over CAN, the other to receive. After connecting the controller boards, there is no communication.... After connecting the ESP32 directly without using the controller board, communication works.... Evidently the problem is in the controller board but I have no idea where to look further for the cause of the problem. All paths lead to the correct pins and have continuity.

stuartpittaway, 2024-04-21:
Just checking that the ESP32 you have is using the DEVKIT-C footprint?  There are several different ESP32 footprint layouts and the pins would be incorrect if its not the DEVKIT-C one.

brii999, 2024-04-21:
I used ESP32-WROOM-32D (ESP32_DevkitC_V4) and now I ordered ESP-32S (both with 38 pins).

Could You recommend some ESP from this page?:
https://allegro.pl/kategoria/plytki-drukowane-moduly-moduly-67363?string=esp32%20devkitc

stuartpittaway, 2024-04-22:
Check your existing ESP32 pin out matches this one... https://components101.com/microcontrollers/esp32-devkitc

brii999, 2024-04-25:
I ordered another ESP32 Devkit C
Still the same result...
I checked if PINs connected to controller board are connected to ESP32 module and they are connected proper PINs of module.

closed 2026-11-14 by stuartpittaway