stuartpittaway/diyBMSv4ESP32 #288: Controller restart when saving config to SD card

jetronic18s, 2024-04-30:
**Describe the bug**
Hello everyone,
I had something really strange yesterday.
I changed something in my settings and wanted to save the configuration to the SD card. 

When I pressed the save button, the controller board rebooted.

I was able to reproduce the whole thing 3 times, it restarted again and again. I then also connected the serial console.

**Hardware/Software Versions**
Controller version (from PCB): 4.5
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: 0f7f03c1c13382ed7031464baa92671fe778afc3
Compiled: 2023-10-30T10:40:07.498Z

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Setting --> Storge'
2. Click on 'Save Config'
4. See error in Logfile 

```
D (72064) diybms: Sent CAN message 0x351
D (72165) diybms: Sent CAN message 0x370
D (72165) diybms: Sent CAN message 0x371
D (72166) diybms: Sent CAN message 0x35e
D (72166) diybms-victron: a8 02 80 00 00 00 00 08 
D (72170) diybms: Sent CAN message 0x35a
D (72173) diybms: Sent CAN message 0x372
D (72180) diybms: Sent CAN message 0x35f
D (72281) diybms: Sent CAN message 0x356
D (72382) diybms: Sent CAN message 0x373
D (72383) diybms: Sent CAN message 0x374
D (72383) diybms: Sent CAN message 0x375
D (72383) diybms: Sent CAN message 0x376
D (72386) diybms: Sent CAN message 0x377
I (72586) diybms: Time now: Mon Apr 29 15:40:13 2024
D (72586) diybms: total_free_byte=91760 total_allocated_byte=198388 largest_free_blk=47092 min_free_byte=76460 alloc_blk=590 free_blk=20 total_blk=610
D (72877) diybms: CANBUS received message ID: 305, DLC: 8, flags: 0
D (72878) diybms: CANBUS received message ID: 307, DLC: 8, flags: 0
I (72953) diybms: Task 1
D (73391) diybms: Sent CAN message 0x351
D (73492) diybms: Sent CAN message 0x370
D (73492) diybms: Sent CAN message 0x371
D (73493) diybms: Sent CAN message 0x35e
D (73495) diybms-victron: a8 02 80 00 00 00 00 08 
D (73497) diybms: Sent CAN message 0x35a
D (73500) diybms: Sent CAN message 0x372
D (73504) diybms: Sent CAN message 0x35f
E (73509) diybms: Unable to get VSPI mutex

assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace: 0x400841a1:0x3ffbc390 0x4008ea2d:0x3ffbc3b0 0x40095419:0x3ffbc3d0 0x4008f652:0x3ffbc500 0x400d7be9:0x3ffbc540 0x400de3a6:0x3ffbc560

ELF file SHA256: ac7daaf9bb5cde74

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    66][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (64) diybms: 

               _          __ 
  _|  o       |_)  |\/|  (_  
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:0f7f03c1c13382ed7031464baa92671fe778afc3 compiled 2023-10-30T10:40:07.498Z
ESP32 Chip model = 1, Rev 3, Cores=2, Features=50
I (236) diybms-hal: Configure I2C
I (237) diybms-hal: Scanning i2c bus
I (241) diybms-hal: Found i2c device at address 0x21
I (250) diybms-hal: Found TCA6416A
I (254) diybms: ** Controller changed state from Unknown to PowerUp **
I (613) diybms-set: nvs_flash_init
I (614) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (614) diybms-hal: CAN driver started
I (624) diybms: LittleFS mounted, total=524288, used=163840
I (751) diybms: Mounting SD card
[   817][W][sd_diskio.cpp:180] sdCommand(): crc error
I (859) diybms-hal: SD card mounted, type 3

Press SPACE BAR to enter console WiFi configuration....

....................

No key press detected
I (5864) diybms-set: Load WIFI config from FLASH - return 1
I (5864) diybms-set: IP=129149120,GW=28485824
I (5864) diybms: Wifi Config: SSID:WaldCafe, Manual Config:1
I (5868) diybms: Manual IP:192.168.178.7, Netmask:255.255.255.0, GW:192.168.178.1, DNS1:192.168.178.1, DNS2:0.0.0.0
I (5879) diybms: Checking for /diybms/wifi.json
D (5887) diybms: Found file /diybms/wifi.json
D (5896) diybms: Deserialized /diybms/wifi.json
I (5897) diybms: Wifi JSON config is identical - ignoring
I (5897) diybms-set: Load config
I (5900) diybms-set: Count: UsedEntries = (245), FreeEntries = (385), AllEntries = (630)
I (5935) curmon: INA229 chip absent
I (5935) diybms: Onboard/internal current monitoring chip not installed
D (5937) diybms: Setup RS485
D (5938) diybms: Configure RS485
D (12378) diybms: CANBUS received message ID: 305, DLC: 8, flags: 0
I (12380) diybms: ** Controller changed state from PowerUp to Stabilizing **
D (12383) diybms: CANBUS received message ID: 307, DLC: 8, flags: 0
D (12392) diybms: starting wifi_init_sta
D (12396) diybms: CANBUS received message ID: 305, DLC: 8, flags: 0
D (12406) diybms: CANBUS received message ID: 307, DLC: 8, flags: 0
D (12411) diybms: CANBUS received message ID: 305, DLC: 8, flags: 0
I (12435) diybms: WIFI SSID: WaldCafe
I (12517) diybms: Hostname: DIYBMS-009FE8F4
D (12518) diybms: wifi_init_sta finished
I (12518) diybms: TFT screen is INSTALLED
D (12521) diybms: total_free_byte=138944 total_allocated_byte=154588 largest_free_blk=98292 min_free_byte=138804 alloc_blk=378 free_blk=5 total_blk=383
I (12523) diybms-tft: Wake up screen
I (13130) diybms: Retry 1, connect to Wifi AP
I (13278) diybms: Request time from time.google.com
I (13285) diybms: Timezone=UTC-1DST-1
I (13290) diybms: The current date/time is: Mon Apr 29 15:40:21 2024
I (13299) diybms-mqtt: Connect MQTT
I (13322) diybms: You can access DIYBMS interface at http://DIYBMS-009FE8F4.local or http://192.168.178.7
D (13322) diybms: CANBUS received message ID: 305, DLC: 8, flags: 0
D (13327) diybms: CANBUS received message ID: 307, DLC: 8, flags: 0
I (13325) diybms-tft: Wake up screen
D (13379) diybms: Sent CAN message 0x351
D (13479) diybms: Sent CAN message 0x370
D (13479) diybms: Sent CAN message 0x371
D (13480) diybms: Sent CAN message 0x35e
```

I know that I'm not on the latest software version, but I did it because I also use a Fritzbox with repeater and mesh. With this version, I have no problems with rebooting when the WLAN is switched off.

Can you do anything with this Stuart?
I've been running the BMS for over a year and have never had this kind of behaviour

stuartpittaway, 2024-04-30:
Right, thanks for reporting this and nice to see the system has been working well for over a year.

I suspect the SD card has become corrupt - if you remove it and reformat it on a PC does it work?  Try another card if you have one spare.

Newer versions of the firmware save the config to flash memory, which is then available for download - so no SD card required.

jetronic18s, 2024-05-03:
Hi Stuart, I inserted a new SD card today. Now everything is working normally again.

I used the old SD card with an adapter, something is wrong with the card. It was only mounted with ReadOnly on my Linux PC. I have not yet carried out any more detailed investigations.

The issue can be closed.

Thanks Stuart

stuartpittaway, 2024-05-03:
Glad it was a simple fix.

closed 2024-05-03 by stuartpittaway