stuartpittaway/diyBMSv4ESP32 #291: Discharge also a curve for amps go down

Michi202020, 2024-06-01:
Hello all, 

I have a question in my charge config I have a charge curve for go down withe the
Amps near the max voltage. Can I also make this near the discharge limit? 
My problem is that the bms say max upload and if the inverter go to max upload 
Amps than jump the voltage to fast down and he stop. This situation I have it a long time before it's empty. 

Best Regards
Michael

Michi202020, 2024-06-25:
Hello Stuart,

for better understanding i have a 70kwh battery withe 3x Deye 12kw hybrid. If i come down to 41v than he have to stop but the discharge current is to high and the voltage go to much down and the system say its to down and i load from the grid to the battery and this go over the night every hour. A Curve for emty the battery is a must to have. see movie below.

https://github.com/stuartpittaway/diyBMSv4ESP32/assets/81771598/49bf3af1-db59-462d-bac3-241390e6e533

Best Regards Michael.G

Michi202020, 2024-08-04:
Hello, No Person have a idea for this?

laserranger3000, 2024-08-28:
So what you want is a de-rating of the current at low SoC (low cell voltage).
I also would like to see this being implemented. A similar logic to the high-SoC de-rating should be sufficient.

Michi202020, 2024-11-10:
Hello people, have you a idea for this situation?
Best regards 
Michael.g

jetronic18s, 2024-11-15:
Hello Michael, 

Do I understand your problem correctly?
You are discharging the batteries with a high current, in the lower SOC range the voltage is then at a value where the inverters report “low voltage”. Then no more current is drawn from the batteries, then the voltage increases again and it starts all over again?

Can you still optimize the cable sizes?
Have you also measured the voltage drop of your connections?

You want a dynamic discharge current control (DCL), similar to the dynamic charging function.

Do I see that correctly?

Can you possibly reduce the voltages on the Deye depending on the current? Victron has something like this, a table with voltage depending on the discharge current.

Regards
Chris

Michi202020, 2024-11-28:
Hi Chris, you're right. My system is already not small with 36kw and 60kwh. My charging current is 500A and discharging current is 700A. Yes, it is so that towards the end the bms says from 700A to 0A and after a few seconds again 700A the voltage jumps and that also goes slower so that a discharge curve is necessary. This will probably not be necessary with small systems with 5 or 10kw, but rather with my large ones. Thank you and best Regards Michael