stuartpittaway/diyBMSv4ESP32 #293: Displayed power is not correct

jetronic18s, 2024-07-15:
I have just noticed that the displayed power is not correct.
62,61V * 0,04A = 2,5 W not 86,25W

[image]

**Hardware/Software Versions**
Controller version (from PCB): 4.4
Processor: ESP32
Version: f210e5be8b0781e24d69c028c44d4f0cadb641d9
Compiled: 2024-05-03T13:10:31.733Z
Language: de

jetronic18s, 2024-07-15:
It must have been the same in the last official release. Huh?

midopple, 2024-07-16:
Hi,

here is an closed Issue for this error

https://github.com/stuartpittaway/diyBMSv4ESP32/issues/240

in the next release this should be fixed

jetronic18s, 2024-07-16:
Thanks for the tip, yes it looks exactly like mine. 
Except that I have an external shunt with INA228.

I am running the latest firmware from Branch “stateofhealth”. there is no fix yet.

midopple, 2024-07-17:
Hi,

I overlooked the fact that an external shunt is used, in this case the values are already calculated in the shunt and transmitted via RS485.

This is why the change in the software does not work here.

the calculation in the firmware of the shunt would have to be adjusted so that this is calculated correctly at low power levels.

I think the code should change like the follow example.
I do not have an external Shunt so i can't test it.

diyBMS-CurrentShunt/Code/diybmsCurrentShunt/src/main.cpp

Line 1160

```
double Power()
{
  
  double busVoltage = BusVoltage();
  double current = Current();
  double pow_val = busVoltage * abs(current); 
  
  if(pow_val < 200){
	return(pow_val);
  }
  else{
    // POWER Power [W] = 3.2 x CURRENT_LSB x POWER
    return (double)i2c_readUint24(INA_REGISTER::POWER) * (double)3.2 * registers.CURRENT_LSB;
  }
}
```

jetronic18s, 2024-07-17:
Thanks, I'll have a look at it and try to compile it.

Unfortunately I don't have direct access to the installation with the external shunt. I might be on site in 6 or 8 weeks.

You would probably have to move the issue to the shunt repo. Only @stuartpittaway can do that

jetronic18s, 2024-07-18:
@midopple 
I have installed your fix, here is the hex file and patch file attached.
[Compiled_Firmware_Workaround.zip](https://github.com/user-attachments/files/16284908/Compiled_Firmware_Workaround.zip)

Would anyone like to test it?