stuartpittaway/diyBMSv4ESP32 #297: Controller crash at wifi dropout

laserranger3000, 2024-09-07:
Hello,

I have just rebooted one of my wifi access points on which the BMS controller was connected to and immediately the controller has crashed and rebooted.
I heard the main relay click, that's how I noticed.
There's another access point with the same SSID and password nearby, so normally the controller should have just connected to that.
I don't know if the disconnection of the old AP or the connection to the new AP caused the crash.

Software: ESP32 2024-08-05

stuartpittaway, 2024-09-08:
Hello, this is a commonly reported issue. Can I ask what WiFi access points do you use?

Are you using a mesh network?

laserranger3000, 2024-09-08:
I'm using two TP-Link Routers running OpenWRT (one Archer C7 AC1750 and one TL-WDR3600). They are connected manually with a 5 GHZ link while the ESP32 is connected to the 2,4GHZ network.
So it's no mesh network.

stuartpittaway, 2024-09-08:
Ok, lots of people had problems with Fritzbox routers when running in mesh configuration.

At the moment I don't have a fix for the WiFi issue, I'm not sure if it's a problem with the esp32 itself (hardware) or something I've done in code - but I've spent weeks debugging that and it looks ok.

It also seems to be variable, for instance if I create a WiFi hotspot on my phone and connect diybms to that I can switch that WiFi on and off all day long without any reboot.

laserranger3000, 2024-09-08:
My two APs are sending on different channels (CH6 and CH11), maybe that could be a reason for the crashes?
I have never experienced this problem when the primary AP rebooted and the controller re-connected to the same AP again. So I guess a simple disconnection is not causing issues, only the re-connection to a different network.

Also, the DHCP server is the same in the whole network, so I would rule out any IP configuration issues.

jetronic18s, 2024-10-25:
@laserranger3000 
A new firmware with the WiFi fix has just been released. Hopefully, fingers crossed!

stuartpittaway, 2024-12-23:
> @laserranger3000 A new firmware with the WiFi fix has just been released. Hopefully, fingers crossed!

How has the testing gone @laserranger3000 ?

laserranger3000, 2025-01-06:
@stuartpittaway:
I have installed Release-2025-01-02-15-31 today.
I could not recreate the issue anymore, so I think it's fixed.

buddhafragt, 2025-01-19:
I Why we dont use a ESP with ethernet and add supoport for this module?

[image]

stuartpittaway, 2025-03-30:
I'm closing this issue as this should now be resolved - please let me know if you still have wifi reboot problems

closed 2025-03-30 by stuartpittaway