stuartpittaway/diyBMSv4ESP32 #300: Adding Wireless AP for WIFI config.

lorenzo620, 2024-11-29:
Hello Stuart,

First, I want to express my gratitude for the incredible work you've done with the DIY BMS project . Your system has been instrumental in effectively managing my energy storage solutions. However, I've identified a couple of areas where the system could be enhanced, and I'd like to propose these improvements for integration into the official release.

**1. Wi-Fi Access Point Mode on First Boot and Network Loss**

One challenge I've encountered is the inability of the DIY BMS to create an access point (AP) on first boot or when the configured wireless network is unavailable. Recently, I changed my wireless router and, without a cable to connect to the DIY BMS, I couldn't update the Wi-Fi credentials. This resulted in a two-day delay until I could purchase a cable and reconfigure the system.

**Proposal:**

Implement a feature where the DIY BMS automatically creates a Wi-Fi access point if it cannot connect to the configured network on boot.
This would allow users to connect directly to the BMS via Wi-Fi to update network settings without the need for a physical connection.

**2. Precharge Relay Functionality**

In many modern Battery Management Systems, there are typically three contactors: one for charging, one for discharging, and another for precharging. Incorporating a precharge function into the DIY BMS would enhance its compatibility with larger and more complex battery systems.

**Proposal:**

Introduce a user-configurable timer for a precharge relay , one of the solid state relays can be used for this. 
Operational Sequence:

**1. Upon power-up, the BMS waits until the system is stable and checks for any triggered alarms.
2. If no alarms are detected, the BMS closes the precharge relay for a user-defined duration (X seconds).
3. After the precharge period, the BMS opens the precharge relay and then closes Relay 1 and Relay 2 for normal operation.**

This functionality is critical for systems utilizing high-capacity batteries and inverters, where precharging ensures safe and reliable startup by preventing inrush currents that could damage components.
In my setup, I use two contactors and an IGBT bridge to control charging and discharging, and this configuration has been operating flawlessly for over a year continuously.

**Additional Notes:**

- Over the past 240 days, I've been using modified firmware by Meningd with intra-CAN communications, data aggregation, and high availability features—all without a single reboot. My system includes three DIY BMS units running in parallel with 4 x 16S packs on each BMS totaling around 180Kwh of LFP energy storage. 
- At a friend's installation, there are six DIY BMS units running in parallel with the same firmware, achieving over 180 days of uptime without faults or reboots. His setup includes seven 16S packs on each BMS, totaling around 600 kWh of LFP energy storage.
- Both my friend and I have extensively tested the modified firmware developed by MENINGD by deliberately triggering various fault scenarios to evaluate how the BMS responds in different conditions. Despite our best efforts to "break" the system, we have not encountered any major flaws in the code. It performs flawlessly, demonstrating stability and reliability in diverse scenarios.
- While we agree that the code could benefit from additional refinement and enhanced safety features, we acknowledge that MENINGD cannot do it all by himself. The foundation he has built is exceptionally solid, and we believe his work deserves recognition and further development to elevate the system's capabilities even more.

Would it be possible for you to collaborate with MENINGD to integrate his modifications and ideas into the official DIY BMS release? His work demonstrates exceptional stability and functionality, and with your expertise, the system could reach new levels of reliability and safety, benefiting the entire DIY BMS community.

Thank you for considering these suggestions. I'm more than willing to assist with testing or provide further details if needed.

pprasse, 2024-12-14:
Hi @lorenzo620,

I suggest a change to the Precharge Relay Functionality.
In my experience with high voltage inverters a precharge functionality using a fixed duration is dangerous. I suggest measuring the output voltage and engaging the main relay if the voltage is high enough (i.e. within a few volts of the expected voltage).
Additionally a make-before-break approach for the main relays is safer.

So I suggest the following sequence:
1. Upon power-up, the BMS waits until the system is stable and checks for any triggered alarms.
2. If no alarms are detected, the BMS closes the precharge relay until the voltage is within X volts of the expected voltage (user-defined)
3. After the precharge period, the BMS closes the relays 1 and 2 (depending on the configured settings)
4. the BMS opens the precharge relay

This sequence can by the way already be used now without any code change if you connect your current/voltage monitor after the precharge circuit.

All the best,

Patrick

lorenzo6201, 2024-12-14:
While the proposed precharge functionality (engaging the main relays based on voltage thresholds) is technically feasible, it may not be the safest or most user-friendly option. Here are some critical points to consider:

Configuration Complexity for Unexperienced Users

Configuring user-defined thresholds requires understanding of the system, and unexperienced users might struggle with this step.
Misconfiguration could lead to unsafe operations or prevent the system from functioning as intended.
To address this, an auto-calibration feature or pre-defined defaults could be implemented to minimize user errors.
Shunt Placement and Voltage Drop Conflict

If the shunt (current/voltage monitor) is placed after the relays, a voltage drop during disconnection (e.g., due to a safety fault) can create conflicting system behavior:
The precharge rule would attempt to re-engage due to the perceived need to recharge the circuit.
Simultaneously, the safety rule would try to keep the relays off, resulting in a rule conflict.
This could lead to oscillations or erratic behavior, compromising system reliability and safety.
Potential Rule Conflicts

In the event of a disconnection triggered by a fault, the system must prioritize safety.
Logic should be implemented to block precharge attempts until the fault is cleared and the system is manually or automatically reset.

pprasse, 2024-12-14:
Side note: I suggest we split the two topics of this issue for the sake of clarity :-)

Regarding faults & safety: Good point. It would make sense to be able to configure the maximum number of "triggers" of a rule before it's considered failed and the fault has to be manually cleared.

Regarding precharge on voltage threshold: This is definitely not the most user-friendly option. But precharging only using timing might work with 48V setups but will get you into very hot water with high voltage setups very quickly.

I currently have a 288V setup and did have the problem of an inverter trying to draw (minimal) power out of the battery during precharging. 
This caused the voltage to reach only ~50V after 30s of precharge. If the main relay had engaged based on timing, the voltage differential between the battery and the inverter capacitors would have been around 230V at that time. Capacitors are like short circuits at the voltage slew rate of a closing relay. 
I have seen things like these at work and trust me, an atomized PCB trace is one of the least dangerous things that happen. On the more problematic side I have seen capacitors disintegrate and burn causing fires, electric arcs that stood there for minutes, etc.

lorenzo6201, 2024-12-14:
In order to clear some misconception. The precharge time you have set at 30 seconds is way to long. Normaly the precharge time is between 1 and 5 seconds. I work daily with HV management system because we are manufacturing HV containers up to 200Mwh and we have management systems that can take 1500V and up to 1000A each and they are up to 15 in paralel. 
So the way that those industrial grade Bms is like this: 
- at first boot it makes some self tests , like reading all cells , making curent leak test , and so on , measuring that they have an inverter at the end of the wires. 
- then is checking that all clusters are at the same voltage if there are more than one in paralel. 
- after all that are passed the precharge relay engages for exactly 5 sec and after that the charge and discharge contactors. 2 seconds after that the precharge contactor opens.

30 seconds of precharge time is way to excesive.

pprasse, 2024-12-15:
Agree, 30sec is not a sensible precharge time. I just wanted to illustrate that after 30sec precharge, voltage was still only 50V.

As you already mentioned in your last post, precharging only by time is not a sensible approach:
A certain voltage must be reached in a certain max time, otherwise it must be considered as a fault.
However, this was not in your initial proposal.

lorenzo6201, 2024-12-15:
The voltage reaching only 50V after 30 seconds occurred because the inverter was powered on and drawing current through the precharge resistor. This approach is not recommended, as the inverter should remain off during the precharge phase. The correct procedure is to power on the inverter only after the precharge process has been fully completed to avoid stressing the precharge resistor and potentially damaging system components.

In industrial or home storage applications, the precharge time is typically fixed at 5 seconds, during which the voltage transitions from 0V to the battery voltage in 3 to 4 seconds. In my procedure, the startup sequence differs slightly. Initially, with the inverter powered off, I activate the Battery Management System (BMS) and wait for the contactors to close. Once the contactors are engaged, I proceed to power up the inverter, followed by the photovoltaic (PV) array and the grid, if available. This method ensures that the inverter does not draw any power through the precharge resistor, as it is off during the precharge phase.

lorenzo6201, 2024-12-15:
Yes, 200 MWh is indeed impressive. The storage system is distributed across 33 40-foot containers, each weighing 80 tons. Each container is equipped with its own Energy Management System (EMS), inverter, and a step-up transformer to convert voltage from 690V to 25,000V or higher. The switchover time between charging and discharging is typically 5 milliseconds. These systems are primarily used for grid balancing and are capable of being fully discharged within 30 minutes.

stuartpittaway, 2024-12-23:
Hi @lorenzo620 is this the same user as @lorenzo6201  ?

Thank you for the suggestions about pre-charge setup and access point options.  They are both good ideas.

I've so far avoided a pre-charge option, generally because everyone has different requirements - personally I use a push button switch and resistor to bypass the main contactor and power up the inverter capacitors.  If my battery or inverter have gone off for any reason - I want to be there to manually reset it/power up - in the past 2 years I've had 1 outage where the fuse breaker tripped.

lorenzo6201, 2024-12-23:
With the precharge should be a usere prefered way to use that relay. As precharge or as output. I would realy love that option. How about the meningd firmware? That firmware rocks.

stuartpittaway, 2024-12-23:
> How about the meningd firmware? That firmware rocks.

I've not personally used it - when it was forked from the original code it was very difficult to re-integrate back into the master branch.  Sounds like you've had success with it though.

lorenzo6201, 2024-12-23:
That firmware is gold. Is worth checking and integrating.

buddhafragt, 2025-01-19:
> That firmware is gold. Is worth checking and integrating.

Hi, where can I download this firmware? I can not find it.