stuartpittaway/diyBMSv4ESP32 #303: Suggestion for new feature

Noob-granit, 2024-12-30:
Hi Stuart

I have been using your DIYBMS for a couple of months now and it´s working Great. 
It´s connected to 3 banks of 16s 300ah LIFEPO4 batteries.
I am however missing a feature, maybe it´s already there and I just haven´t been able to find it but I would like to be able to control the relay outputs based on SOC. Is this possible, or could it be an improvement?

What I am trying to achieve with this is to control an external relay to turn on and off the AC input to the inverter, this because it has a rather high idle consumption that I could do without when the battery is in a high state of charge. 

Best regards
Tobbe

stuartpittaway, 2025-01-02:
Hi, good to hear the system is working well.

There isn't a rule related to state of charge - you may be able to use the "bank over voltage" rule to emulate this - but that may not be suitable with LIFEPO4 cell voltages unless at the extremes of charging/discharging.

red0909, 2025-01-07:
> There isn't a rule related to state of charge 

can you add thet?