stuartpittaway/diyBMSv4ESP32 #305: Communication 128 cells

Leodien, 2025-02-16:
Stuart,

here are the images, hopefully.

[image]
[image]
[image]

Leodien, 2025-02-19:
Construction

[image]

Leodien, 2025-05-01:
Stuart,

Below are the print-screen images made with a 1 bank, 128 cell configuration.

[image]
[image]
[image]
[image]

Leodien, 2025-05-01:
Stuart
Last image is with an 8-bank, 16-cell configuration.

Leodien, 2025-05-01:
I have also added a log file of the 1 bank and 128 module configuration

[25-05-01-W (32456) diybms-mqtt MQTT enabled,.txt](https://github.com/user-attachments/files/19998818/25-05-01-W.32456.diybms-mqtt.MQTT.enabled.txt)

stuartpittaway, 2025-05-01:
Ok great.

What I'm seeing is the out of sequence error counts being more than zero.  This indicates that communication is getting lost in the cell modules and probably when you have gaps in the charts.

What communication speed (baud rate) are you using?

Leodien, 2025-05-01:
10k

stuartpittaway, 2025-05-01:
Try the 5K option.  I think the problem you have is lost data rather than not supporting 128 cells.  You shouldn't have missing gaps in the chart, which is why the waiting for modules error is showing

stuartpittaway, 2025-05-01:
You can also take a look at the module page and look at the bad packet counts.  Send a screenshot of that over.

stuartpittaway, 2025-05-09:
Dropped you a message on Patreon, if you can upload some more screenshots please

Leodien, 2025-07-24:
Stuart,

I'm doing some testing and noticed that the BMS is giving CAN register 4220 "Charge Cutoff Voltage" the value 0.
This is causing my inverter to display a "high battery voltage" error.
Is this due to a setting, or is it hard-wired?

Thank you.

Leodien, 2025-07-27:
Stuart, thanks for the reply.

Indeed, I'm using the Pyloforce HV protocol.

If I:

if (rules.IsChargeAllowed(&mysettings))
{
data.battery_charge_voltage = rules.DynamicChargeVoltage();
data.battery_charge_current_limit = 30000 + (uint16_t)max((int16_t)0,rules.DynamicChargeCurrent());
}
else
{
ESP_LOGV(TAG, "Charging not allowed in message 4220");
}

Replace with:

data.battery_charge_voltage = rules.DynamicChargeVoltage();

Does the value remain active?