stuartpittaway/diyBMSv4ESP32 #306: DIYBMS stop working

buddhafragt, 2025-02-24:
**Describe the bug**
A clear and concise description of what the bug is.

**Hardware/Software Versions**
Controller version (from PCB): ??
(below can be obtained from the "About" page in the controller web interface)
Processor: ??
Version: ???
Compiled: ??

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

buddhafragt, 2025-02-24:
Hello,
I use the last firmware and notice that the MPPTs in Venus do not feed in from time to time. (0A). DIYBMS also gives me a message that the feed-in is switched off. in Charge/discharge configuration “Prevent charging ” is not checked, 
if I then switch it on once and switch it off again, everything works again. 
Here is someting wrong.....

jetronic18s, 2025-03-14:
Do you mean the kind of behavior described here? 
Then this has nothing to do with the DIYBMS.

https://community.victronenergy.com/t/ess-mit-dvcc-mppt-regelt-mittags-kurz-auf-null/9447

jetronic18s, 2025-03-22:
I seem to have solved this problem for me.

I have done the following:
1. deactivate DC feedin
2. multiplus --> advanced --> recognize VE bus system again
3. reactivate DC feedin

Since I did this, it hasn't happened again. So 4 days now...

https://community.victronenergy.com/t/mppt-leistung-geht-auf-0-wenn-die-batterie-voll-ist-sollte-aber-uberschuss-einspeisen/20985/45

Can you test this for yourself?

buddhafragt, 2025-04-08:
Hello Jetronic,
I don't have a Multiplus, so no VE-Bus!
When I reload DIYBMS in the browser, I see a warning from DIYBMS ‘charging is prevented’. So I think it's the DIYBMS. 
In the meantime I have changed the hardware, updated the software, checked the settings 10 times, but the same error. Soon I will replace DIYBMS, too much trouble. 

> I seem to have solved this problem for me.
>
> I have done the following:
>
>     1. deactivate DC feedin
>
>     2. multiplus --> advanced --> recognize VE bus system again
>
>     3. reactivate DC feedin [...]

red0909, 2025-05-03:
> Soon I will replace DIYBMS, too much trouble.

i have now the jk inverter bms in new battery packs, no problems and very good balancing.
but the problem all other bms have is the bad soc calculation, this is the only reason the diy bms is still in use at my system.
i cant use the victron shunt because i need the precise soc transmitted to my sma sunny island.
no other "cheap" bms can do this and after the last wifi problem fix fw i had never a issue again. it is rockstable with 2x 16s balance boards and external shunt pcb.