stuartpittaway/diyBMSv4ESP32 #307: Controller prevents charge and discharge via CANBUS

jmonsted, 2025-03-08:
**Describe the bug**
With the DIYBMS connected to a Deye 12k inverter, data about the battery and BMS gets correctly reported, but the charge and discharge limits are set to 0A. I can't seem to find any reason why this would be the case. I copied the settings from one of stuart's other responses here, as i am also running a 16S LFP battery (2P16S of EVE 314Ah cells) just tweaked with the 200A limit on my design. I can switch the inverter over to "voltage" mode and it all works as expected. I checked the serial output from the controller for any anomalous messages, but didn't see anything interesting. 

Controller, 16S module and passive balancer purchased from stuart's shop. Voltage measurements don't agree with my multimeters, but are at least within the limits i've configured. That'll be my next thing to try to correct.

**Hardware/Software Versions**
Controller version (from PCB): BETA-0b93100
(below can be obtained from the "About" page in the controller web interface)
Processor: ESP32
Version: be7e21c65bf65c3d4854afff583b05c4d052e574
Compiled: 2025-01-02T15:29:45.705Z

Relay output 1 hooked into a shunt trip that pops the breaker and disconnects the supply from the inverter.

[image]

[image]

[image]

[image]

buddhafragt, 2025-04-18:
for me the same, Controller prevents charge, its is a bug. I already mentioned it here a long time ago.

stuartpittaway, 2025-05-26:
Ok, randomly disappearing temperature values and also discharge prevented messages are a likely indicator that communication messages are being lost and the controller assumes "zero" voltage or "no temperature" for those missed messages.

You should have a ZERO value in the tiles for OOS Errors, CRC Errors and Ignored Requests.   If they are not zero, reset the counters (utilities page) and see if they increase over a couple of hours.

In regards to the MaxCharging & MaxDischarging are they ever non-zero? DIYBMS emulates a Pylontech battery using the 0x351 CANBUS message which transmits these 4 values...

```
    uint16_t battery_charge_voltage;
    // positive number
    int16_t battery_charge_current_limit;
    // negative number
    int16_t battery_discharge_current_limit;
    uint16_t battery_discharge_voltage;
```

I believe those values are what you see in the "Charge Current Limit" on the display - you should be able to test that by changing the values in DIYBMS and they should update.

Now I suspect the DEYE inverter is expecting another data packet with the Max Charge/Discharge values in it.  I don't have a DEYE inverter so no way to test this.  Is there any documentation on the DEYE inverter and CANBUS communication?

stuartpittaway, 2025-05-26:
Just found this on Google... [Sol-Ark CAN Bus Protocol V1.3.pdf](https://github.com/user-attachments/files/20442602/Sol-Ark.CAN.Bus.Protocol.V1.3.pdf)

jmonsted, 2025-05-27:
I've added some debug messages to the firmware to see what gets sent:

I (816912) diybms-pylon: Sending PYLON CANBUS 351: 549, 1538, 2000, 488
I (816992) diybms-pylon: Sending PYLON CANBUS 35c: 192

jmonsted, 2025-05-27:
I notice i've been getting weird fluctuations in the voltages. I guarantee this 628 Ah cell hasn't jumped .1V while it was idle. 
I wonder if that could play into anything?

[image]

stuartpittaway, 2025-05-27:
Did you check for communication errors?

How are the cells attached to the monitoring board?

Have you tried selecting non-deye inverter option on the charge screen?

jmonsted, 2025-05-28:
I see no communication errors.

[image]

Cells are connected with 5x1.5 mm2 cable and deutsch connectors, matched cable lengths. Overkill is the best kind of kill :)

Switched inverter brand to Generic with no change. 

Just noticed that it does appear to periodically work. I left it on lithium-mode (i've usually switched it back to voltage mode to let the batteries actually work) and it has charged and discharged at some times.

[image]

jmonsted, 2025-05-29:
https://www.deyeinverter.com/deyeinverter/2024/01/04/deyeapprovedbatterylistdy-lv48-0076.pdf doesn't mention victron, but i haven't been able to find a list of what the battery codes mean. I've tried both pylon over can (mode 00) and pylon over rs485 (mode 12) with the same result.

I haven't found a way to see why the charge/discharge rules are blocking. Would i have to add debug messages to the code?

I tried "Override State of Charge" just now  with no discernible difference. Same by toggling "prevent charge" and "prevent discharge" on and off. I don't see "force charge" anywhere.