stuartpittaway/diyBMSv4ESP32 #310: Add option to prioritize reserve capacity (hold float mode indefinitely)

menningd, 2025-03-23:
In some installations such as mobile applications there is a preference towards maintaining a fully (or mostly fully) charged battery, and expecting charge sources to offset any parasitic loads while in this state. Installations like these can have the battery sitting for extended periods (weeks or even months) of time whilst expected to maintain preparedness for a full discharge cycle. Presently the DIYBMS firmware is tailored more to ESS's where daily charge/discharge cycles are expected. The result of using the current firmware in an application such as, e.g. a boat thats kept plugged in to shore power, is that the parasitic loads will continously short cycle the battery between 100% and 95% (or whatever the SOC resume charge is set to) every couple of days. 

As a further enhancement to "charge modes" there should be an option to disable the float timers expiration so that it stays in float mode indefinitely. Then the battery can be kept at a high SOC and the charger will handle supplying parasitic loads without trying to force anymore charge into the cells. Of course  it is vitaly important to set the correct float voltage for the cell chemistry. And the battery will have dropped down to 95-98% SOC once it reaches this float voltage. But this is the best compromise for these types of applications.

stuartpittaway, 2025-03-23:
Isn't this already possible by lowering the "S.o.C charge resume" to a very low value?  Float is maintained for up to 720 minutes at the moment, but that should be easy to expand.  You would probably want the battery to top up say every 7 days, even if running on a boat.

menningd, 2025-03-23:
Lowering 'SOC resume charge' isn't a great option because you're almost guaranteed _not_ to have full charge when you show up expecting it. Alternatively I dont think cycling between 100-95 SOC every other day is better for the battery then just letting it sit *idle for a couple weeks at a time. 
*(Of course the DC system isn't idle - just zero current in/out of the cells. In these applications it doesn't make sense to cycle the batteries with these smaller loads when there is power available via the grid or solar)

But I do agree with you increasing the timer is the best solution (and also the easiest). Because ultimately you still need to occasionally trigger a rebalancing. I would think up to 3-4 weeks is still reasonable to wait to trigger a reabsorption (rebalancing) phase for an unused battery. It looks like Victrons built in charge profile defaults to every 7 days reabsoption phase for LFP and is configurable.
 "Mysettings->floatvoltagetimer" is already a uint16 so we could just open up the upper limit. Even better would be to add **Days** + Minutes so one doesn't have to specify 30,240 minutes instead of 3 weeks. Still backwards compatible with existing user settings that are specified in minutes only (default to 0 days).

menningd, 2025-03-23:
Lowering 'SOC resume charge' isn't a great option because you're almost guaranteed _not_ to have full charge when you show up expecting it. Alternatively I dont think cycling between 100-95 SOC every other day is better for the battery then just letting it sit *idle for a couple weeks at a time. 
*(Of course the DC system isn't idle - just zero current in/out of the cells. In these applications it doesn't make sense to cycle the batteries with these loads when there is power available via the grid or solar)

But I do agree with you increasing the timer is the best solution (and also the easiest). Because ultimately you still need to occasionally trigger a rebalancing. I would think up to 3-4 weeks is still reasonable to wait to trigger a reabsorption (rebalancing) phase for an unused battery. It looks like Victrons built in charge profile defaults to every 7 days reabsoption phase for LFP and is configurable.
 "Mysettings->floatvoltagetimer" is already a uint16 so we could just open up the upper limit. Even better would be to add **Days** + Minutes so one doesn't have to specify 30,240 minutes instead of 3 weeks. Still backwards compatible with existing user settings that are specified in minutes only (default to 0 days).

stuartpittaway, 2025-03-24:
The BMS will drop out of float mode if the soc is lower than the SOC resume value - ignoring the float timer.

If we want to add days/hours then that's a UI change, we can leave the internals using a minute variable.

menningd, 2025-03-24:
> If we want to add days/hours then that's a UI change, we can leave the internals using a minute variable.

I think day/hours would be perfect. I'm not sure if anybody would miss the granularity of minutes since the default is 6 hours. Just one point of reference Victron specifies only in days [https://www.victronenergy.com/media/pg/VEConfigure_Manual/en/charger-settings.html](url). Here they're referring to a repeated absorption time (since these settings are also meant to be configured for lead acid) where with diyBMS we're just talking about going to "Charge Stopped" and then letting the SOC fall in order to trigger a new charge cycle.

closed 2026-11-14 by stuartpittaway