stuartpittaway/diyBMSv4ESP32 #313: Numbers are missing in Home View

olterion, 2025-04-02:
I made no update for long time. Today I uploaded the [Release-2025-03-30-15-36](https://github.com/stuartpittaway/diyBMSv4ESP32/releases/tag/Tag-2025-03-30-15-36) and there are no numbers at the bars on home view.

Processor: ESP32
Version: [e6db6e77a2f627bfb19121321a74e1887f0af646](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/e6db6e77a2f627bfb19121321a74e1887f0af646)
Compiled: 2025-03-30T15:34:39.989Z

Is this a bug or a feeature?
For me personally, I want to have the numbers at the bars.

[image]

olterion, 2025-04-04:
Yes, I made several refreshes with Ctrl+F5, I tried also an other device and an other browser.
Noting helps, everywhere the same view.
If I upload the old firmware, everything is ok.

stuartpittaway, 2025-04-07:
Hi @olterion this was actually a requested change way back from March 2024 as users complained that they couldn't read the numbers/chart when you had more than 24 cells on screen.

https://github.com/stuartpittaway/diyBMSv4ESP32/commit/d67f47683581b05d481d473741630fba35ffaba4

stuartpittaway, 2026-11-14:
Added in 0379124 on branch `March2026-MinorChanges`.

Settings → Display Settings now has a **Graph bar labels** option:
- **Automatic (hide above 24 cells)**: the default, the same as the current behaviour from #270
- **Always show**: labels on every bar, whatever the number of cells
- **Never show**

Choose **Always show** to keep the numbers on the bars with more than 24 cells. The graph updates as soon as the setting is saved, and the setting is included in settings backup/restore.

closed 2026-11-14 by stuartpittaway