stuartpittaway/diyBMSv4ESP32 #316: Victron canbus "diybms: Failed to queue CANBUS message (0x107)"

dougnutz, 2025-05-01:
I'm trying to troubleshoot an error with a new install (existing bms)
I have an [Ekrano GX](https://www.victronenergy.com/media/pg/Ekrano_GX/en/index-en.html) that I am trying to connect to the ve.Canbus using the DIYbms canbus integration. I was hoping this would work in the same fashion as the cerbo gx as they are both GX units.
 I am getting the following error when I enable the can bus.
`diybms: Failed to queue CANBUS message (0x107)`

no messages seem to reach the Ekrano GX. 
I have verified the canbuss baud rate at both ends as 250k
I have verified both ends are properly terminated with 120ohm resistor (verified with fluke)
I double checked my wiring for the canbus Hi/low. I tried a few different settings and nothing seems to change anything. 
Looking at the Ekrano it does not register any packets incoming. Dropped or errors. 

I have a similar build running on my home install with a Sol-Ark inverter with the DiyBms emulating the victron bus. It has worked flawlessly for over a year. So I have reasonable confidence in the build/hardware. But seem to be missing something. 

I am not sure where to start next, maybe ssh into the Ekrano and see if there are any packets coming in?

I was using a stable build that I compiled myself but tried the "latest" release just to be sure it wasn't on my end
Processor: ESP32
DIYBMS V4 hardware 

Version: [e6db6e77a2f627bfb19121321a74e1887f0af646](https://github.com/stuartpittaway/diyBMSv4ESP32/commit/e6db6e77a2f627bfb19121321a74e1887f0af646)

Compiled: 2025-03-30T15:34:39.989Z

```E (138846) diybms: Failed to queue CANBUS message (0x107)
I (138846) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:49561, state: RUNNING
I (139798) diybms-webreq: API call: monitor2
E (140104) diybms: Failed to queue CANBUS message (0x107)
I (140104) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:51926, state: RUNNING
E (140458) diybms: Failed to queue CANBUS message (0x107)
I (140458) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:52591, state: RUNNING
E (140712) diybms: Failed to queue CANBUS message (0x107)
I (140712) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:53069, state: RUNNING
E (140966) diybms: Failed to queue CANBUS message (0x107)
I (140966) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:53546, state: RUNNING
D (141142) diybms: Task 3, s=0 e=13
E (141220) diybms: Failed to queue CANBUS message (0x107)
I (141220) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:54024, state: RUNNING
E (141474) diybms: Failed to queue CANBUS message (0x107)
I (141477) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:54507, state: RUNNING
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    62][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (77) diybms:

               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:e6db6e77a2f627bfb19121321a74e1887f0af646 compiled 2025-03-30T15:34:39.989Z
ESP32 Chip model = 1, Rev 3, Cores=2, Features=50
I (238) diybms-hal: Configure I2C
I (239) diybms-hal: Scanning i2c bus
I (243) diybms-hal: Found i2c device at address 0x21
I (252) diybms-hal: Found TCA6416A
I (255) diybms: ** Controller changed state from Unknown to PowerUp **
I (612) diybms-set: nvs_flash_init
I (616) diybms: LittleFS mounted, total=524288, used=24576
I (632) diybms: Mounting SD card
[  1194][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1199][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  1210][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[... 85 lines cut ...]
E (20794) diybms: Failed to queue CANBUS message (0x107)
I (20794) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:9596, state: RUNNING
E (21048) diybms: Failed to queue CANBUS message (0x107)
I (21048) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:10056, state: RUNNING
E (21302) diybms: Failed to queue CANBUS message (0x107)
I (21302) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:10516, state: RUNNING
E (21556) diybms: Failed to queue CANBUS message (0x107)
I (21556) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:10976, state: RUNNING
E (21810) diybms: Failed to queue CANBUS message (0x107)
I (21810) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:11436, state: RUNNING
E (23164) diybms: Failed to queue CANBUS message (0x107)
I (23164) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:13889, state: RUNNING
E (23518) diybms: Failed to queue CANBUS message (0x107)
I (23518) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:14530, state: RUNNING
I (23532) diybms-rules: Set error 4:WaitingForModulesToReply
I (23533) diybms-rules: Set warning 9:NoExternalTempSensor
I (23533) diybms-rules: Set warning 7:ChargePrevented
I (23537) diybms-rules: Set warning 8:DischargePrevented
I (23542) diybms: Active errors=1
E (23772) diybms: Failed to queue CANBUS message (0x107)
I (23772) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:14991, state: RUNNING
E (24026) diybms: Failed to queue CANBUS message (0x107)
I (24026) diybms: CAN STATUS: rx-q:0, tx-q:6, rx-err:0, tx-err:128, arb-lost:0, bus-err:15451, state: RUNNING

```

stuartpittaway, 2025-05-01:
These error codes can be decoded bus-err:10516

They are reported by the esp32 hardware itself. See if you can find a reference on the web for that.

stuartpittaway, 2025-05-01:
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/can.html#alerts

dougnutz, 2025-05-01:
Thanks @stuartpittaway I managed to sort myself out on the communication issue with the help of those links. Now have packets flowing, both the diybms and ekrano gx are registering packets sent/received. The numbers correlate properly so data seems to be flowing.

However, I'm not seeing any of the battery information show up on the Ekrano GX screen. No SOC, temp etc.
The device list on the GX doesn't show the DIYBMS on the main devices page or the canbus devices like it does on your instruction videos. 

Do I need to enable something on the GX?
GX

[image]

[image]

[image]

[image]

[image]

stuartpittaway, 2025-05-01:
I've not used that particular device but I would have expected it to just show up without any specific configuration.

dougnutz, 2025-05-01:
Ok so enabling nmea200-out did the trick the diybms showed in the list and was automatically selected by the system.

[image]

closed 2025-05-01 by dougnutz